titanous/homebrew-gnuradio #55: installing gnuradio on 10.10.1

mmonka, 2015-01-09:
i try to install gnuradio on my mac (10.10.1)

> gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.0 (clang-600.0.56) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.0.0
Thread model: posix

$ brew install gnuradio
==> Installing gnuradio from titanous/homebrew-gnuradio
==> Downloading http://gnuradio.org/releases/gnuradio/gnuradio-3.6.5.1.tar.gz
######################################################################## 100,0%
==> Patching
patching file grc/CMakeLists.txt
patching file gr-qtgui/lib/spectrumdisplayform.ui
==> cmake .. -DCMAKE_PREFIX_PATH=/usr/local/Cellar/gnuradio/3.6.5.1 -DQWT_INCLUDE_DIRS=/usr/local/lib/qwt.framework/Headers -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gnuradio/3.6.5.1 -DCMAKE_BUILD_TYPE=None -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DENABLE_GR_QTGUI=OFF -DENABLE_DOXYGEN=OFF 
==> make
ld: -rpath can only be used when targeting Mac OS X 10.5 or later
collect2: ld returned 1 exit status
make[2]: *** [volk/lib/libvolk.0.0.0.dylib] Error 1
make[1]: *** [volk/lib/CMakeFiles/volk.dir/all] Error 2
make: *** [all] Error 2
couldn't understand kern.osversion `14.0.0'

READ THIS: http://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/homebrew):
  https://github.com/titanous/homebrew-gnuradio/issues

Any idea how to solve this? Do you need more details for debugging?

freelancer77, 2015-01-18:
I second that.

Oxtay, 2015-01-22:
You might've already tried this, but according to [this link](http://stackoverflow.com/questions/26486163/gcc-4-9-1-in-os-x-yosemite-gcc-warning-couldn-t-understand-kern-osversion-1), the solution might be to upgrate `gcc`:

`brew upgrade gcc`

wyager, 2015-02-03:
Same here. Upgrading gcc did not help. I also upgraded to 10.10.2, and the message changed from "couldn't understand kern.osversion `14.0.0'" to "couldn't understand kern.osversion `14.1.0'".

galgolan, 2015-02-06:
Im also having the same issue:
Gals-MacBook-Air:~ galgo$ brew install gnuradio --with-qt
==> Installing gnuradio from titanous/homebrew-gnuradio
==> Downloading http://gnuradio.org/releases/gnuradio/gnuradio-3.6.5.1.tar.gz
Already downloaded: /Library/Caches/Homebrew/gnuradio-3.6.5.1.tar.gz
==> Patching
patching file grc/CMakeLists.txt
patching file gr-qtgui/lib/spectrumdisplayform.ui
==> cmake .. -DCMAKE_PREFIX_PATH=/usr/local/Cellar/gnuradio/3.6.5.1 -DQWT_INCLUD
==> make
ld: -rpath can only be used when targeting Mac OS X 10.5 or later
collect2: ld returned 1 exit status
make[2]: *** [volk/lib/libvolk.0.0.0.dylib] Error 1
make[1]: *** [volk/lib/CMakeFiles/volk.dir/all] Error 2
make: *** [all] Error 2
couldn't understand kern.osversion `14.1.0'
Warning: It appears you have MacPorts or Fink installed.
Software installed with other package managers causes known problems for
Homebrew. If a formula fails to build, uninstall MacPorts/Fink and try again.

miaoski, 2015-02-25:
Same thing here
```
==> cmake .. -DCMAKE_PREFIX_PATH=/usr/local/Cellar/gnuradio/3.6.5.1 -DQWT_INCLUDE_DIRS=/usr/local/lib/qwt.framework/Headers -DCMAKE_IN
==> make
make[2]: *** [volk/lib/CMakeFiles/volk.dir/volk_16i_s32f_deinterleave_32f_x2_a_orc_impl.c.o] Error 1
make[2]: *** [volk/lib/CMakeFiles/volk.dir/volk_16ic_deinterleave_real_8i_a_orc_impl.c.o] Error 1
make[2]: *** [volk/lib/CMakeFiles/volk.dir/volk_16ic_magnitude_16i_a_orc_impl.c.o] Error 1
make[1]: *** [volk/lib/CMakeFiles/volk.dir/all] Error 2
make: *** [all] Error 2
couldn't understand kern.osversion `14.1.0'
```

propellerhat, 2015-03-07:
I, too, have this issue.

Firstly, I installed apple-gcc42 because it was listed as a dependency:

    brew install --with-qt gnuradio
    ==> Installing gnuradio from titanous/homebrew-gnuradio
    Error: No available formula for apple-gcc42 (dependency of gnuradio)
    Searching formulae...
    Searching taps...
    homebrew/dupes/apple-gcc42

So I installed apple-gcc42 like it wanted:

    brew install homebrew/dupes/apple-gcc42
    Cloning into '/usr/local/Library/Taps/homebrew/homebrew-dupes'...
    remote: Counting objects: 1581, done.
    ...

Then the installation of gnuradio proceeded until I had the same error as mmonka:

    brew install --with-qt gnuradio
    ==> Installing gnuradio from titanous/homebrew-gnuradio 
    ...
    <Several lines omitted>
    ...
    ==> Installing gnuradio
    ==> Downloading http://gnuradio.org/releases/gnuradio/gnuradio-3.6.5.1.tar.gz
    ######################################################################## 100.0%
    ==> Patching
    patching file grc/CMakeLists.txt
    patching file gr-qtgui/lib/spectrumdisplayform.ui
    ==> cmake .. -DCMAKE_PREFIX_PATH=/usr/local/Cellar/gnuradio/3.6.5.1 -DQWT_INCLUDE_DIRS=/usr/local/lib/qwt.framework/Headers -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gnuradio/3.6.5.1 -DCMAKE_BUILD_TYPE=None -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_V
    ==> make
    ld: -rpath can only be used when targeting Mac OS X 10.5 or later
    collect2: ld returned 1 exit status
    make[2]: *** [volk/lib/libvolk.0.0.0.dylib] Error 1
    make[1]: *** [volk/lib/CMakeFiles/volk.dir/all] Error 2
    make: *** [all] Error 2
    couldn't understand kern.osversion `14.1.0'
    
    READ THIS: http://git.io/brew-troubleshooting
    If reporting this issue please do so at (not Homebrew/homebrew):
      https://github.com/titanous/homebrew-gnuradio/issues

This seems to be an error generated by the version of gcc it wanted. If I run the apple-gcc42 that was installed, I get the same error:

    gcc-4.2
    couldn't understand kern.osversion `14.1.0'
    i686-apple-darwin11-gcc-4.2.1: no input files

Lines 49-51 in gnuradio.rb force compilation with gcc-4.2. I'm going to attempt to change that to gcc-4.9 and see if it works. I'll report back soon...

red-green, 2015-07-28:
Any news? I have this problem too...
I tried uninstalling macports as suggested, and it broke the funcionality of `ld`, but I fixed that by setting MACOSX_DEPLOYMENT_TARGET=10.9. The error came back.

andrewmoskal, 2015-08-28:
[  6%] Linking C shared library libvolk.dylib
cd /tmp/gnuradio20150828-82063-1bgzvwj/gnuradio-3.6.5.1/build/volk/lib && /usr/local/Cellar/cmake/3.3.1/bin/cmake -E cmake_link_script CMakeFiles/volk.dir/link.txt --verbose=1
/usr/local/bin/gcc-4.2   -dynamiclib -Wl,-headerpad_max_install_names  -o libvolk.0.0.0.dylib -install_name /tmp/gnuradio20150828-82063-1bgzvwj/gnuradio-3.6.5.1/build/volk/lib/libvolk.0.0.0.dylib CMakeFiles/volk.dir/volk_16i_s32f_deinterleave_32f_x2_a_orc_impl.c.o CMakeFiles/volk.dir/volk_16ic_deinterleave_16i_x2_a_orc_impl.c.o CMakeFiles/volk.dir/volk_16ic_deinterleave_real_8i_a_orc_impl.c.o CMakeFiles/volk.dir/volk_16ic_magnitude_16i_a_orc_impl.c.o CMakeFiles/volk.dir/volk_16sc_magnitude_32f_aligned16_orc_impl.c.o CMakeFiles/volk.dir/volk_16u_byteswap_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_s32f_multiply_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_s32f_normalize_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_sqrt_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_add_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_divide_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_dot_prod_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_max_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_min_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_multiply_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32f_x2_subtract_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32fc_32f_multiply_32fc_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32fc_magnitude_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32fc_s32f_magnitude_16i_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32fc_s32fc_multiply_32fc_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32fc_x2_multiply_32fc_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32i_x2_and_32i_a_orc_impl.c.o CMakeFiles/volk.dir/volk_32i_x2_or_32i_a_orc_impl.c.o CMakeFiles/volk.dir/volk_8i_convert_16i_a_orc_impl.c.o CMakeFiles/volk.dir/volk_8i_s32f_convert_32f_a_orc_impl.c.o CMakeFiles/volk.dir/volk_prefs.c.o CMakeFiles/volk.dir/volk_rank_archs.c.o CMakeFiles/volk.dir/volk.c.o CMakeFiles/volk.dir/volk_cpu.c.o CMakeFiles/volk.dir/volk_machines.c.o CMakeFiles/volk.dir/volk_machine_generic_orc.c.o CMakeFiles/volk.dir/volk_machine_sse2_64_mmx_orc.c.o CMakeFiles/volk.dir/volk_machine_sse3_64_orc.c.o CMakeFiles/volk.dir/volk_machine_ssse3_64_orc.c.o CMakeFiles/volk.dir/volk_machine_sse4_1_64_orc.c.o  -L/usr/local/Cellar/orc/0.4.23/lib /usr/local/lib/liborc-0.4.dylib -Wl,-rpath,/usr/local/Cellar/orc/0.4.23/lib 
couldn't understand kern.osversion `14.5.0'
ld: -rpath can only be used when targeting Mac OS X 10.5 or later
collect2: ld returned 1 exit status
make[2]: *** [volk/lib/libvolk.0.0.0.dylib] Error 1
make[1]: *** [volk/lib/CMakeFiles/volk.dir/all] Error 2
make: *** [all] Error 2
couldn't understand kern.osversion `14.5.0'
/usr/local/Library/Homebrew/debrew.rb:11:in `raise'
BuildError: Failed executing: make

andrewmoskal, 2015-08-28:
brew install -vd --with-qt gnuradio

andrewmoskal, 2015-08-28:
In case HEAD version:

Your branch is up-to-date with 'origin/master'.
git reset --hard origin/master
HEAD is now at 22e2f1a cmake: update revision to 3.7.8 for feature release
git submodule foreach --recursive git submodule sync
Entering 'volk'
git submodule update --init --recursive
==> Checking out branch master
==> Patching
patching file grc/CMakeLists.txt
patching file gr-qtgui/lib/spectrumdisplayform.ui
Hunk #1 FAILED at 518.
1 out of 1 hunk FAILED -- saving rejects to file gr-qtgui/lib/spectrumdisplayform.ui.rej
/usr/local/Library/Homebrew/debrew.rb:11:in `raise'
ErrorDuringExecution: Failure while executing: /usr/bin/patch -g 0 -f -p1