titanous/homebrew-gnuradio #58: Any help on this error!

senadaruc, 2016-08-31:
Dears,

I am receiving the following error when i am ruining the brew upgrade:

saruc@SARUC-M-K22S:~|⇒  brew upgrade
Warning: Calling MacOS.xcode_version is deprecated!
Use MacOS::Xcode.version instead.
/usr/local/Library/Taps/titanous/homebrew-gnuradio/rtlsdr.rb:13:in `<class:Rtlsdr>'
Please report this to the titanous/gnuradio tap!

Cheers,
S

rixon, 2017-04-19:
Thanks for reporting.  That's a warning, not an error, meaning it will not block installation.  It's safe to ignore that when building (for now.)

Please use a more helpful topic when reporting issues.  For example, using "Warning: Calling MacOS.xcode_version is deprecated" would be a good subject for this issue, allowing everyone to have an idea of what the issue is before opening it.

kiddyfurby, 2017-06-22:
$ brew tap titanous/homebrew-gnuradio
Error: Another active Homebrew update process is already in progress.
Please wait for it to finish or terminate it to continue.
==> Tapping titanous/gnuradio
Cloning into '/usr/local/Homebrew/Library/Taps/titanous/homebrew-gnuradio'...
remote: Counting objects: 7, done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 7 (delta 0), reused 2 (delta 0), pack-reused 0
Unpacking objects: 100% (7/7), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/titanous/homebrew-gnuradio/gnuradio.rb
Calling Formula.sha1 is disabled!
Use Formula.sha256 instead.
/usr/local/Homebrew/Library/Taps/titanous/homebrew-gnuradio/gnuradio.rb:6:in `<class:Gnuradio>'
Please report this to the titanous/gnuradio tap!
Warning: Calling MacOS.xcode_version is deprecated!
Use MacOS::Xcode.version instead.
/usr/local/Homebrew/Library/Taps/titanous/homebrew-gnuradio/rtlsdr.rb:13:in `<class:Rtlsdr>'
Please report this to the titanous/gnuradio tap!

Error: Cannot tap titanous/gnuradio: invalid syntax in tap!

Its actually blocking?